Ultimaker/Cura #10545: Skirt/Brim Speed Overrides First Layer Speed

SquishinStix, 2021-09-30:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Prusa Mini+

### Reproduction steps

Enable supports and a full brim for those supports. Enable a draft shield for something less than the full extent of the model. Set the "Initial Layer Speed" to 20mm/s. Set the "Skirt/Brim Speed" to 60mm/s.

### Actual results

The first layer of the supports and draft shield will print at 60mm/s, despite the "Initial Layer Speed" being set at 20mm/s. 

### Expected results

The entire initial layer, including the supports and draft shield, should print at the "Initial Layer Speed" of 20mm/s. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have a file with supports enabled and brim enabled for those supports. I also have a draft shield enabled for about half the height of the model (10mm of 20mm). My "Initial Layer Speed"  is set to 20mm/s and my "Skirt/Brim Speed" is set to 60mm/s since it's printing for half the height of the model. The "Skirt/Brim Speed"  overrides the "Initial Layer Speed" for the first layer causing the first layer of the supports and draft shield to print way too quick and therefore not adhere to the bed. My workaround was to just lower the "Skirt/Brim Speed" to 20mm/s but since the entire draft shield is then printed at that speed, it added about 10% to my overall print time.

Fix is to have the "Initial Layer Speed" override ANY AND ALL other speed settings for the very first layer. I can't imagine a situation where you would want your skirt and/or brim to print faster or slower than the rest of the print on the first layer.

fvrmr, 2021-10-04:
Hi @SquishinStix thank you for your report. Could you share your project file (file - save project) and share it as a zip here. In that way it is easier to reproduce your issue. Thank you in advance

SquishinStix, 2021-10-04:
Yes! I figured out how to work around the issue, but I still think it is an
issue. In the attached file you will see that the "Skirt/Brim" speed of
40mm/s is overriding the "Initial Layer Speed" of 15mm/s. I want my entire
first layer nice and slow but because I am using a brim on my supports, the
first layer of my draft shield and supports are printing at 40mm/s. I
worked around this by lowering the skirt/brim speed which fixed my supports
first layer but also slowed down the subsequent layers of the draft shield.
It wasn't a huge deal, workaround only added a couple minutes, but it was
hard to figure out what was overriding the "Initial Layer Speed" and I
think it should override *any* and *all* other speed settings.

By the way, I started with PrusaSlicer and quickly wanted more flexibility
and Cura has been *awesome *to use (not that PrusaSlicer is bad, just not
as flexible) - thank you for the great product!

Thank you!

fvrmr, 2021-10-05:
Good to know that you've found a work around. I still want to take a look at your file. But I can't see your attached file, since you've replied via email. You need to post it in a comment on Github here: #10545

SquishinStix, 2021-10-05:
Oops - sorry! First time using GitHub and didn't even notice that the email notification I got wasn't a direct email. I have attached the file below (had to .zip it for it to attach).

[Initial Layer Speed Issue.3mf.zip](https://github.com/Ultimaker/Cura/files/7287475/Initial.Layer.Speed.Issue.3mf.zip)

closed 2021-10-07 by Ghostkeeper

SquishinStix, 2021-10-07:
Sorry you couldn't check my file, but everything you stated is exactly correct. Having a separate draft shield speed as opposed to having it match the skirt/brim speed OR having the support brim speed set separate from the primary part brim speed would fix this issue.
Thanks!

aleph00, 2026-09-24:
Hi guys !
This issue is quite old, but I ran into the same problem lately and didn't find any improvement on it since 2021. So, with the help of chatgpt as I'm not a developper, we wrote a post-processing script to change the speed of the draft (and only that). You can find it on my github : [Cura-DraftShieldSpeed](https://github.com/aleph00/Cura-DraftShieldSpeed)