Ultimaker/Cura #10546: delay extruder heating till after UBL mesh test has completed

jxer, 2021-09-30:
### Is your feature request related to a problem?

nozzle and bed heat up at same time. UBL with bl touch takes a couple minutes, this causes filament to ooze, while waiting for 
 UBL test to finish. 

### Describe the solution you'd like

can we get an option to set extrude temp or delay it, its added before the custom gcode on a ender3v2 

### Describe alternatives you've considered

n/a

### Affected users and/or printers

ender 3v2

GregValiant, 2021-09-30:
The answer is "Yes".  Using "Keywords" in the Start Gcode gets noticed by Cura and so it doesn't add those first temperature lines.

An Ender has enough power supply to heat the bed and hot end at the same time.  This is the stock Start Gcode for an Ender 3 and because there are no Keywords in regards to temperature - Cura adds the temperatures at the very start.
; Ender 3 Custom Start G-code
G92 E0 ; Reset Extruder
G28 ; Home all axes
;G29 ;if there is ABL then the command usually goes here.
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish

And I think something like this is what you are looking for...
; Ender 3 Custom Start G-code
G92 E0 ; Reset Extruder
M104 S170 ;Warm up the hot end for leveling but too cold to ooze.  Don't wait.
M190 S{material_bed_temperature_layer_0} ;Heat the bed and wait for it to hit the operating temperature.
M109 S170 ;Wait just in case the hot end didn't get to 170 while the bed was heating.
G28 ; Home all axes
G29 ;AutoLevel
M109 S{material_print_temperature_layer_0} ;Heat the hot end up to printing temperature and wait for it to get there.
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z10.0 F3000 ; Move Z Axis up 10 to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
;End of StartUp Gcode

jxer, 2021-09-30:
@GregValiant 

Pretty much perfect. Ty for taking the time.

closed 2021-09-30 by jxer