Ultimaker/Cura #10548: Marketplace unavailable from within Cura

pnt103, 2021-10-01:
### Application Version

4.9.0, 4.10.0

### Platform

Windows 10 Pro 64-bit, incl all current updates

### Printer

Ender 3 Pro

### Reproduction steps

After a new install of Cura 4.10.0 or 4.9.0 on a new PC, running Windows 10 Pro 64-bit, which has no previous Cura install, I clicked on Marketplace to install plugins.  All I get is an almost-blank window stating "Could not connect to the Cura Package database. Please check your connection"."  This appears immediately, without the usual delay in filling the Marketplace window.  Whether I'm logged in to Ultimaker or not, I can see the installed plugins but cannot select or view the _Plugins_ or _Materials_ tabs.

### Actual results

I always get the window with the error, which doesn't change even if I leave it open for ten minutes, or close and re-open it.  The same thing happens with Cura 4.10.0 and 4.9.0.  The same happens on my old PC, which was working fine last time used, a couple of weeks ago, with both 4.9.0 and 4.10.0.

### Expected results

I should see a selection of plugins after a moderate delay, and  be able to select tabs for functional plugins and materials.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I've tried a clean install of both 4.9.0 and 4.10.0 on the "new" PC, as well as installs after copying all the files from a previously-working installation on another PC.  Nothing on my local network has changed.  Can this be related to the announcement about problems with Let's Encrypt SSL certificates, at http://status.ultimaker.com/ ?

pnt103, 2021-10-01:
I forgot to mention that I have also tried with Windows Firewall and my antivirus s/w temporarily disabled, and checked that Windows Defender and my AV haven't blocked anything.

fieldOfView, 2021-10-01:
This cannot be diagnosed without logs.

pnt103, 2021-10-01:
Never mind.  It turns out it was because of the Let's Encrypt root certificate issue.  Not the fault of Ultimaker, Cura, the marketplace, or even Let's Encrypt:  recent new installations of Windows 10 (I tested three Pro and one Enterprise) have the old DST Root CA X3 installed but not the newer ISRG Root X1.  Given that's been available for six years, it's just stupidity on the part of a certain OS vendor.  Anyway, after installing the new certificate and removing the old one, the Marketplace works again.

closed 2021-10-01 by pnt103

nallath, 2021-10-01:
Yes, it's related to that issue. It seems that windows hasn't pushed all certs. When i tried this morning it didn't work. After I updated windows it did work.

pnt103, 2021-10-01:
It didn't work for me on three new (in the last two weeks) installs that also have all updates from Microsoft.  But removing the old SSL root certificate and installing the new one fixed it.

nallath, 2021-10-01:
It might not be related to the update. Someone did point me to this tweet: https://twitter.com/hmoffatt/status/1443848907893592072.

pnt103, 2021-10-01:
Yes, I saw that.  I downloaded the "new" (only six years old) root cert (isrgrootx1.der) from Let's Encrypt (https://letsencrypt.org/certificates/), installed with certlm.msc, deleted the old X3 cert, and restarted Cura.  It now works.

Dim3nsioneer, 2021-10-05:
> 
> 
> Yes, I saw that. I downloaded the "new" (only six years old) root cert (isrgrootx1.der) from Let's Encrypt (https://letsencrypt.org/certificates/), installed with certlm.msc, deleted the old X3 cert, and restarted Cura. It now works.

Also used this workaround successfully with Cura 4.11