Ultimaker/Cura #10549: Cura Settings Guide

AKCirc, 2021-10-01:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Ultimaker

### Reproduction steps

Clicked on Infill - connect infill lines. Requested help information.

### Actual results

 In Cura Settings Guide 2.7.1  the images for Connect Infil Lines are identical. The description states that these should be very different.

### Expected results

The images should be different.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

All projects have this result.
[image]
)

nallath, 2021-10-01:
Please report issues with a plugin with the repository of the plugin. In this case that would be: https://github.com/ghostkeeper/settingsguide

closed 2021-10-01 by nallath

Ghostkeeper, 2021-10-08:
You can see in your own screenshot that in the left picture, the lines have open ends, while in the right picture the endpoints of the lines are connected via the side ;) Just needs a more careful eye, is all.

Maybe reducing the scale of that model could make it more clear to see.