Ultimaker/Cura #10550: Error slicing

inqusi, 2021-10-01:
### Application Version

4.11.0

### Platform

WINDOWS 10

### Printer

Creality Ender 3

### Reproduction steps

Opened the model STL, slicing
[23.zip](https://github.com/Ultimaker/Cura/files/7269451/23.zip)

### Actual results

error
[image]

### Expected results

slicing

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CE3_23.zip](https://github.com/Ultimaker/Cura/files/7269460/CE3_23.zip
[cura.log](https://github.com/Ultimaker/Cura/files/7269473/cura.log)
)

TheHawk25, 2021-10-03:
Check #10418, many people seem to have this problem if you use zigzag patterns with the new mono-direction slicing.

Festivejelly, 2021-10-03:
I've noticed I get this if I do "arrange all models" but only on my Tronxy X5SA profile, works fine on CR10. If I manually place them on my tronxy it seems to work fine.

closed 2021-10-04 by fvrmr