Ultimaker/Cura #10551: Failure to slice with some models

TheHawk25, 2021-10-02:
### Application Version

4.11

### Platform

Windows 10

### Printer

Ender 5 Pro (Modified)

### Reproduction steps

I opened up the collapsing broadsword model from here: https://www.thingiverse.com/thing:3591373 and when attempting to slice I get the error 
'Slicing failed
Slicing failed with unexpected error. Please consider reporting a bug....'
I have since attempted to change settings such as the location of the model or its orientation, in addition I have also reinstalled cura in an attempt to resolve the issue. I have had the same issue with other models such as this figure: http://www.thingiverse.com/thing:2887306 , but oddly only the male model is affected, the female versions seem to slice just fine.
During slicing it often takes roughly 2x as long as normal and often gets stuck just before halfway before it reaches the end and shows the error.

### Actual results

I got an error after a long time waiting to slice stating that the slice had failed due to an unexpected error and to report the bug here.

I have placed the project files of the broadsword and figurine in the zip files below

### Expected results

The model should have just sliced normally, in the case of the figurine models, I'd expect that the male model works just as well as the female one, but obviously that isn't the case. I'm sure there must be other models that cause this issue, however given that these slices where it fails often take 2x as long, I don't want to go through my model archive to figure out which ones cause the errors.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[CE5_COLLAPSING_BROADSWORD.zip](https://github.com/Ultimaker/Cura/files/7271228/CE5_COLLAPSING_BROADSWORD.zip)
[CE5_man.zip](https://github.com/Ultimaker/Cura/files/7271229/CE5_man.zip)

TheHawk25, 2021-10-02:
I've just noticed that rarely, rotating the object will allow it to slice again, but rotating it back leaves me stuck again.

smartavionics, 2021-10-02:
Possible duplicate of #10418 ?

TheHawk25, 2021-10-03:
Yes, this does appear to be the same problem mentioned in #10418 and the fixes/workarounds there seem to resolve my issue.

closed 2021-10-03 by TheHawk25