Ultimaker/Cura #10556: Option to add custom printer is missing

bigbeau, 2021-10-03:
### Application Version

4.11

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

1. Opened cura 4.11, updated from an older version (4.9)
2. Tried to add a custom printer

### Actual results

There is no option under non-networked printers to add or configure a custom printer. The option is gone.

### Expected results

I expect to see the custom tab, and within it, the option to create a new FFM printer.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7273995/cura.log)
[image]

fieldOfView, 2021-10-03:
It looks like you have "optimised" the Custom DDD Printer away with the Start Optimiser plugin. In `Extensions -> Start Optimiser`, select the option `Restore all configuration files`. Then restart Cura. This will restore all printer definitions, including the Custom FFF Printer.

The `Custom FFF Printer` option is under `Add a non-networked printer`, `Custom`.

bigbeau, 2021-10-03:
> It looks like you have "optimised" the Custom DDD Printer away with the Start Optimiser plugin. In `Extensions -> Start Optimiser`, select the option `Restore all configuration files`. Then restart [...]

Oops! Yeah, that was it. Thanks, haha

closed 2021-10-03 by bigbeau