Ultimaker/Cura #10561: Dual Extrusion: On every other layer I only get a monochrome E0 layer - E1 is ignored

ScruffR, 2021-10-04:
### Application Version

4.11

### Platform

Windows 10 Pro 64 bit

### Printer

Modded Creality CR-10 with E3D Chimera Dual Extruder

### Reproduction steps

I tried to create a two-color label with a monochrome (E0) initial layer for best bed adhesion in Fusion 360. 
Thereafter there should be the background as E0 with embedded text coming from E1. 
In order to have the text as one single object I added a top layer of 0.01mm which would be omitted by Cura.

[Labels.zip](https://github.com/Ultimaker/Cura/files/7276990/Labels.zip)
*Contains `Labels.3mf` (Fusion 360 export) and `CFFFP_Labels.3mf` (Cura project file)*

### Actual results

In preview, for some reason, I get alternating layers of E0 ***only*** extrusion followed by a combination as intended (E0 background plus E1 text).

Correct layer 3
[image]

Incorrect layer 4
[image]

### Expected results

All layers after layer 1 should be looking like layer 3.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

nothing to add

ScruffR, 2021-10-04:
Duh 🤦‍♂️ !

I found my error - I had forgotten to boolean-cut the text from the back plate.

closed 2021-10-04 by ScruffR

Ghostkeeper, 2021-10-11:
An alternative solution would be to disable the "Alternate Mesh Removal" setting.