Ultimaker/Cura #10565: Memory usage skyrockets with landscape model!

Zogar89, 2021-10-04:
### Application Version

4.11

### Platform

Windows 10

### Printer

Artillery X1

### Reproduction steps

Load Project.
Slice Project.

Slicing freeze in the midle and memory usage skyrockets! 

After a few seconds the program delivers an error message and suggests reporting it here

### Actual results

.

### Expected results

.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7281742/cura.log
[14-15.zip](https://github.com/Ultimaker/Cura/files/7281743/14-15.zip)
)

fvrmr, 2021-10-05:
Hi @Zogar89 thank you for your report.
When opening your model I noticed a lot of watertight errors. Please try to fix your model first and try again when your model is fixed.

GregValiant, 2021-10-09:
780,812 non-watertight edges
 32,838 reversed faces

Is that Atlantis?

Ghostkeeper, 2021-10-12:
This is actually another instance of this bug: https://github.com/Ultimaker/Cura/issues/10418

As a workaround, disable Monotonic Ironing. Peak memory usage for me was 532MB then.