Ultimaker/Cura #10567: Curves Not Slicing

crumhirnd, 2021-10-05:
### Application Version

4.11.0

### Platform

MacOS 11.6

### Printer

Custom Printer

### Reproduction steps

1. Import model
2. Click "Slice"

### Actual results

The model is not sliced, only a small support is made

### Expected results

The model is sliced according to user settings

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[TOP.zip](https://github.com/Ultimaker/Cura/files/7282128/TOP.zip)

GregValiant, 2021-10-05:
For such a little model that one sure has a lot of errors.  When the "Model is not watertight" warning comes up it is a red flag in regards to the model.  I uploaded the model to an online STL repair site and got this report before it timed out and refused to repair the model:

-> Reading file and indexing vertices
-> Analysed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> **1144 Non-manifold edges (?)**
--> **1589 Inverted faces (?)**
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)

I loaded the model into MS 3D Builder and it took a while but seems to have fixed the model.  Try the attached version.  You may notice that the file size went from 10.8kb to 35.7kb.
My take on it is that much of the model was inside-out.  BTW it has 732034 faces and 365810 vertices.

[GVTopRep.zip](https://github.com/Ultimaker/Cura/files/7285399/GVTopRep.zip)

crumhirnd, 2021-10-05:
I'm not a pro, I just modelled it in fusion and exported as an obj and then
stl when that didn't work.

GregValiant, 2021-10-05:
Some models that aren't "Watertight" have minor problems that Cura can blast through.  Others (like this one) are beyond what Cura can deal with.  Here the repaired model is on the left.
[image]

I learned something today.  This is the X-Ray view and the color difference is due to the errors in the model.  Now I have a better idea of what to look for.
[image]

crumhirnd, 2021-10-05:
Thanks for the tips and of course the repair, but again I am not a
professional

GregValiant, 2021-10-06:
_"I am not a professional"_

Me neither, but we will be.  These machines are at the same time simple (it goes back-and-forth and squirts plastic) and complicated (software, hardware, firmware, gcode, plugins, model design, CAD software, and on and on).