Ultimaker/Cura #10569: slicing trouble

LTHxxx, 2021-10-05:
### Application Version

4.11.0

### Platform

WINDOWS 10

### Printer

Creality Ender 3 Pro

### Reproduction steps

slicing .stl from https://www.thingiverse.com/thing:548230 with 0.4 nozzle
fin is not sliced, see 3dPrintPlane_1.png
when rotating -10° fin is only partially sliced, see 3dPrintPlane_2_-10°.png

all is ok with Slic3r

### Actual results

parts not sliced

### Expected results

sliced 100%

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[3dPrintPlane.zip](https://github.com/Ultimaker/Cura/files/7283577/3dPrintPlane.zip)

fvrmr, 2021-10-05:
Hi @LTHxxx thank you for your report. 
It looks like the fin is too thin. You could enable `print thin walls` or use our Arachne version.

Thin walls enabled:
[image]

Arachne:
[image]

Read more about Arachne here: https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_beta

Let me know if that works for you!

LTHxxx, 2021-10-05:
that's perfect, thank's a lot !

closed 2021-10-05 by LTHxxx