Ultimaker/Cura #10571: Missing "Normally Calculated" Symbol for modified Retraction Settings

seanatwood8, 2021-10-05:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Custom FFF Printer

### Reproduction steps

1. Change retraction distance or retraction speed setting
2. Update Profile

### Actual results

1. "Retraction Distance" or "Retraction Speed" becomes italicized

### Expected results

1. Function symbol (fx) should appear next to the retraction distance setting indicating "This setting is normally calculated, but currently has an absolute value set."

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Retraction settings can come from the material settings OR the profile settings, and it isn't clear which it's being pulled from because it doesn't show the function symbol, as is done with other settings, such as the printing temperature, which can also be calculated from the profile OR the material settings. This is inconsistent!

Please make the retraction settings show the function symbol in a consistent manner as the printing temperature! Both of these settings can be controlled by the material OR the profile settings, so they should work the same!
[image]
[image]

fvrmr, 2021-10-08:
Hi @seanatwood8 thank your report. The retraction distance doesn't affect the retraction speed. 

Changing the retraction speed will affect the retraction retract speed, retraction prime speed and wipe retraction speed. So after changing the retraction speed you want to change the retraction retract speed you will see the fx icon. Because these settings are in the defined in the profile settings.
[image]

Let me know if this makes sense to you.

Ghostkeeper, 2021-10-15:
The fx button is planned to be replaced by something that is simpler to understand anyway, so we're not planning to make that button itself more clear. I'm glad I could clear up the confusion in here, but the real solution will come from that change.

closed 2021-10-15 by Ghostkeeper