Ultimaker/Cura #10572: Option for first layer(s) to not draw isolated walls first

alandsidel, 2021-10-05:
### Is your feature request related to a problem?

We often use brims to help adhesion of prints that have sharp corners, so the corners don't detach during the print.  When printing a part without a solid base, the gcode cura creates will print the brim first, then isolated interior wall sections, then wall sections connected to the brim (I may have these two out of order).  The interior "floor" is printed last, connecting the exterior wall to the interior walls that surround voids, which is the issue.

The problem is that these interior walls often have sharp angles, and they are printed without a brim, so they can begin curling from the build surface before the printer has connected them to the exterior walls via the "floor" layers.

In the included image you can see what I'm talking about.  The print itself is not yet failing in the photo, but occasionally it does, because the "keyhole" shapes and sharp corners of the rectangle above them are not connected to anything and are being left to cool for a rather long time before the first "floor" layer is printed to connect them to the rest of the print, and they are not printed with a brim of their own to ensure adhesion.

### Describe the solution you'd like

I would like to see another option in cura along the lines of "print interior walls last" so that the print order on a part like the one in the photo would look something like this:

1. Print brim first (if brim enabled).
2. Print as much of the layer as possible that is connected to the brim, avoiding "void" areas.
3. Print the interior walls separating the already printed material from the voids last.

This will ensure that the majority of the first layer is printed together, minimizing the time that small printed areas remain isolated to cool.

An alternative to this would be an option to print some of the first layer around the interior walls before those walls are printed, acting as a brim for these isolated regions.

### Describe alternatives you've considered

The only existing alternative I can think of to solve this issue is to print with a brim instead of a raft.  This does work, but it wastes material and results in an "ugly" first layer when the print is complete.

Another alternative was mentioned in the previous section.

### Affected users and/or printers

Anyone printing items that have intentional gaps / holes with sharp angles in the first layer of their print will benefit from this.

### Additional information & file uploads

[image]

Ghostkeeper, 2021-10-12:
So effectively what you mean is that some part of the bottom skin of the model is printed in concentric fashion rather than the normal skin pattern, and that this pattern gets optimised together with the walls then. Or, similarly, a setting to adjust the number of walls on the initial layer?

Fribb, 2022-09-28:
I would like to add that this isn't only for holes in the bottom layer but is also a problem with the printed wall order.

For example, you would want to set your wall ordering to "inside to outside" for the model so that you have something to hold onto and that your outer wall is looking better.

But this ordering is also applied to the Bottom layer, the brim is printed first, then the inside wall and then the outside wall which means that the inside wall is printed without connection to the brim which is more likely to fail.

A workaround at the moment is to add a support blocker and create a cutting mesh overlap at the size of the first layer that is then setting the wall order to "outside to inside" just for the first layer. The image below shows the problem on the XYZ Calibration cube.

My point is, whatever is printed on the first layer should prioritise being connected to already printed parts.

[image]

githubDiversity, 2023-11-06:
I'd like to second this feature request. I just can get those inner walls to stick to the build plate. But a brim always does. So an inner brim that overlaps with the inner walls that are printed later would be a solution for me at least.

But the original requested solution would work best.