Ultimaker/Cura #10573: glitch and wrong printing time and percentage

giorgioskizzo, 2021-10-05:
### Application Version

4.11

### Platform

Windows 10

### Printer

Anet A8 customized with 50x25cm printbed and marlin

### Reproduction steps

1. load stl in cura
2. generate gcode
3. saved it on disk
4. closed the program (to print via sd, but printer don't liked the memory card)
5. open cura
6. load gcode from memory card
7. print it via usb

### Actual results

the software calculated 26 hours of printing (1 hour from slicing calculation)
the percentage is wrong and negative

see the video, the time clock don't go regular, but accelerate and slow down continuously, depending of what layer section the printer is printing.

Bonus:
there is an overlapping on small windows sizes

### Expected results

correct time and percentage

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

https://user-images.githubusercontent.com/11254832/136057050-8805669b-65fc-411c-bd27-b93a84ca9a80.mp4
[AA8_screw2.zip](https://github.com/Ultimaker/Cura/files/7287298/AA8_screw2.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/7287304/cura.log)

giorgioskizzo, 2021-10-05:
after some minutes...
[image]

fvrmr, 2021-10-08:
Hi @giorgioskizzo thank you for your report. The calculations are based on your printer settings. And this is something we don't maintain since we are depended on our community for adding these printer profiles and can't test this.
However the -23% is something I haven't seen before. Could you share a project file (file - save project). You gcode doesn't have your settings.

About the overlap, was that only during the screen capture?

Ghostkeeper, 2021-10-12:
We do not give USB printing any priority since none of Ultimaker's printers support USB printing. It is a bug, but don't expect it to be fixed any time soon. Sorry!

closed 2023-06-09 by github-actions[bot]