Ultimaker/Cura #10574: Prime tower only primes extruder 1 on layer 1

opy001, 2021-10-05:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

I have 6 extruders.  I turned on "enable prime tower" and expected it to print that with the appropriate extruder before printing the model in that specific color.  

### Actual results

The first layer prints the entire first layer of the prime tower with extruder 1 then prints the rest of the colors with no priming.  On layer 2 it actually primes each extruder just fine.

### Expected results

Layer 1 prime tower should prime each extruder before actually using it to print the model.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CFFFP_PokerChip-inlays.zip](https://github.com/Ultimaker/Cura/files/7287658/CFFFP_PokerChip-inlays.zip)

fvrmr, 2021-10-08:
Hi @opy001 thank you for your report.
The reason why the first layer is with one extruder is for better adhesion. A work around is to enable the `prime tower brim`. If you enable this you should change the `brim width`. 
Let me know if that works for you!

opy001, 2021-10-08:
@fvrmr I added the brim and it puts x number of lines on the brim for the specified extruder 3, then it puts 1 brim line of extruder 2 (the only 2 I am using for this specific test, I have 6 total) then the entire rest of the inner tower is extruder 1 which should never be used on this print.  Then every layer above that primes extruder 1 as well as the other extruders, which extruder 1 is never used so I'm not sure why it is priming.

While this would solve a 2 color issue on layer one if I used extruder 1 as one of those, if I use 3 or more colors it does not prime the third extruder enough.

I am working with custom towers until this can be fixed.  I thought it worked correctly on 4.8.  I was doing this last year, but had to take a break for family health reasons.

Even with the prime tower issues, I am getting WAY better prints now than I was with S3D.  The Cura team has done an incredible job on this slicer and it has now become my primary slicer again.

TheSkittle, 2021-10-11:
@opy001 I am having another problem with prime towers. heres the link: https://github.com/Ultimaker/Cura/issues/10602

opy001, 2021-10-12:
@TheSkittle Im actually using my own towers now.  I have a diamond 5 color and my custom towers allows me to purge all 5 nozzles at once then retract and fill the void with my new color to prevent bleeding of other colors.  It is working quite well so far.

Ghostkeeper, 2021-10-12:
This is indeed expected behaviour.

So the need to prime is not exclusive to multi-extruder printers. Even single-extruder printers need to prime at the start of the print (and those prints don't use prime towers). Cura provides an option for that: the skirt. Or alternatively, the brim. If these are not used, the printer is expected to prime for themselves at the start of the print, for instance during the start g-code to produce a line, or by using the G280 prime blob command.

The prime tower is exclusive to multi-extruder prints. However it doesn't assume that using a prime tower means that the extruders suddenly stop priming at the start of the print. Cura's options for the skirt and the brim add a line around the skirt/brim for each extruder, so that all extruders get primed the same way. In your print you've set the Adhesion Type to None, so you're expected to have the printer prime *all extruders* in the start g-code. If this is cumbersome for your printer, then I suggest using a skirt.

Aside from that, the prime tower needs to stick well to the build plate since it's rather tall and thin. Since different materials usually don't stick well to each other nor stick equally well to the build plate, the material with the greatest adhesion tendency is chosen for the entire first layer of the prime tower. This same material is also used for the outer shell of the prime tower, which is printed on every layer to ensure that the prime tower doesn't delaminate. This also answers your later question:

> Not sure why extruder 1 is priming on ever single layer when it is never used.

So the material with greatest adhesion tendency (or the first such material in case of ties) is used as the outer shell of the prime tower to ensure that the layer adhesion is good enough to make the prime tower not break halfway during the print.

Can you let me know if using a skirt is an acceptable solution for you? If not, we'd like to know why not and what we can do in the start g-code.

TheSkittle, 2021-10-12:
@Ghostkeeper i also turned z hop on. Do u think thats what caused this?

opy001, 2021-10-12:
@TheSkittle Can you print the cone with one color on that printer?

TheSkittle, 2021-10-14:
> @TheSkittle Up to you, but I use anywhere from 50-90mm depending on the color and the amount of bleeding you get.

ok. I'll try it. Why not just use a regular prime tower though?

opy001, 2021-10-14:
> ok. I'll try it. Why not just use a regular prime tower though?

Because my nozzle has all 5 colors in it at one time and I need to purge all 5 then retract and fill the space with the next color to prevent bleeding from another port.  So my prime tower start by extruding x amount of extruder 6 (virtual extruder using all 5 at once) then it retracts 15 mm and the next color purges 15mm to fill in the void and finishes priming on the tower.  Each color change does this to reduce the chance of bleeding.

TheSkittle, 2021-10-14:
https://www.amazon.com/gp/product/B08X4MS655/ref=ox_sc_act_title_1?smid=A3ND1AQ2DDBI35&psc=1 this is the link

opy001, 2021-10-15:
@Ghostkeeper This issue is fixed as far as I am concerned.  Enabling shared extruder fixed a lot of my issues.