Ultimaker/Cura #10575: I need to reset Configurations in Cura too often

aedwardd, 2021-10-05:
### Application Version

4.11.0

### Platform

Mac 11.6

### Printer

Ultimaker S5

### Reproduction steps

I had this problem starting about 4 months ago. Its as if the center point for the work space isn't calibrated right in Cura. I researched the problem and found resetting the Configurations worked fine. Set up my profiles again and it work for a couple of weeks. Reset configurations again and it happened within a couple of days. Tried one more time and same thing, a couple of days with it working properly. Cura is still usable but it makes it difficult to move or duplicate anything as the only way I can select it is through the 'Object list'. Images show how Cura opens up for me, focused on the lower left corner of the work bed, and the other image is when I select the object, the arrows become larger than usual and they aren't selectable with the cursor.

### Actual results

Works for a couple weeks to a couple days.

### Expected results

Works all the time

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]
)

aedwardd, 2021-10-08:
thanks @fvrmr

[cura.log](https://github.com/Ultimaker/Cura/files/7313346/cura.log)

fieldOfView, 2021-10-10:
Do you have multiple displays?

It may matter which display Cura starts up on. Resetting the Cura configuration will also reset which display the application opens on, but it is a very blunt way to achieve just that. 

On the General pane of the Cura preferences, please uncheck the "Restore window position on start" checkbox. This way, Cura will always start up on the primary display.