Ultimaker/Cura #10577: "Add Printer" not working

CHolstein25, 2021-10-05:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Creality CR-6 SE

### Reproduction steps

1. Clicked on "Add printer" in the printers menu
2. Went to "Non-networked printers" list

### Actual results

Only the printers that I already have added to my Cura page are showing up.

### Expected results

I should've seen a list of printers in the Cura database to choose from
[image]

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

<img width="361" alt="CuraProblem1" src="https://user-images.githubusercontent.com/91992922/136105665-8526a51e-5300-4787-885e-3ff7a0860e2c.png"
[image]
>

nallath, 2021-10-06:
Do you have the startup optimizer plugin enabled perhaps? 

If not; Could you share your log files?

fieldOfView, 2021-10-07:
> Only the printers that I already have added to my Cura page are showing up.

Note that this is *exactly* what startup optimiser means to do; it does not load the definitions that you don't use. You can undo the optimisations via the Extensions -> Start optimiser menu, add a new printer of a type you did not use before, and then redo the optimisation. The plugin will then "keep" all the files for the printers you have configured and not load the others.