Ultimaker/Cura #10579: [4.11] Cooling turns on at 100% when ironing

printingotb, 2021-10-06:
### Application Version

4.11.0

### Platform

Win 10

### Printer

custom fff, repetier firmware

### Reproduction steps

Load my project, slice and search for M106.

### Actual results

M106 S255 is inserted when it starts ironing the final layer.

### Expected results

I can't think of any combination of settings that should cause this. 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[Cooling bug project.zip](https://github.com/Ultimaker/Cura/files/7294164/Cooling.bug.project.zip)
[Cooling bug.gcode.zip](https://github.com/Ultimaker/Cura/files/7294165/Cooling.bug.gcode.zip)

nallath, 2021-10-06:
This has been fixed for 4.12 :) 

https://github.com/Ultimaker/CuraEngine/commit/304704251947e47d42171b9a344cc04d0dd0e004

Original report: #10407

printingotb, 2021-10-06:
Great! 
Forgot to search the closed issues 🤫