Ultimaker/Cura #10594: Cura UI bugs out after waking from sleep on linux

bhwarren, 2021-10-10:
### Application Version

4.11.0

### Platform

Linux (Elementary OS 6 based on Ubuntu 20.04)

### Printer

Voron 2 350

### Reproduction steps

1. Launch cura and slice a model
2. let the computer go to sleep
3. wakes from sleep and the cura ui graphics are now buggy

### Actual results

The ui graphics look bugged out. The ui itself is still usable, for example if you hover over  the menu items, it will then display normally and the buttons are functional, albeit without the proper visual graphics.

### Expected results

The UI should look exactly as it does before the computer sleeps.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This is on Elementary OS 6, which is based on Ubuntu 20.04. I'm using the nvidia driver, version 470 with a 1070 graphics card. This is the only application I use that presents this behavior

[image]

bhwarren, 2021-10-12:
Thanks for the response! I did not realize the snap package was unofficial, sorry about that. I downloaded the official AppImage, and it is producing similar results after going to sleep though
[image]

Ghostkeeper, 2021-10-18:
I'm unable to reproduce this problem using Ubuntu 21.04, with NVidia RTX2070 driver version 460. After waking up from sleep, and unlocking my lock screen, Cura still looks normal.

Maybe we'll have to see if other users also encounter this issue and look for what is common among the people that have it.

Ghostkeeper, 2021-10-19:
Updated my Ubuntu installation to 21.10 yesterday and the drivers to version 470, but after waking up from sleep it's still fine.

masterxq, 2021-12-16:
I have the same problem with AppImage 4.12.1.

Using:
- Debian Testing

NVIDIA Driver:
- 470.86-1

Using KDE desktop environment

So it is most likely not repaired with updating the NVIDIA Driver... I have some other programs that have very similar problems. It looks like they never refresh some content stored in the Video RAM that was without supply for a while (sleep). After sleep the my cura looks very much like on the screenshot in the bug report.

It would be really nice to have this fixed, it crashes my workflow after a break^^

jimduchek, 2022-03-29:
I just verified that this issue does NOT exist w/AMDGPU drivers on an AMD APU (no dedicated VRAM), with either suspend or hibernate.  And was able to reproduce easily on my other machine with a 1080ti.  The machines both have more-or-less identical installations of Archlinux (same QT, same Cura install, etc).  Am curious if someone with a dedicated Radeon card can reproduce it. 

I strongly suspect that this is an upstream bug in QT, though.

MariMakes, 2023-10-19:
Quick update from our side 👋

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, and we've introduced a lot of changes since this report.  I'll be closing this issue. I hope you understand. 

Apologies for any inconvenience caused by our delayed response. 
Thanks again, and please let us know if you run into any other issues 💪

closed 2023-10-19 by MariMakes