Ultimaker/Cura #10611: Add support for Material Type to allow +, silk and Noctilucent

bodyboarder2528, 2021-10-13:
### Is your feature request related to a problem?

So when making my own Materials so I can work out cost and have different print temps, I made a PLA+ one but noticed it said "not supported". I also noticed the same when I added Silk and Noctilucent in the material type field. When I removed those then it would be fine. The issue is its not just PLA and we should be able to put the actual material type in there to limit confusion when needing to change the material that I am using when slicing a job.

### Describe the solution you'd like

The ability to use +, Noctilucent and silk to the material type without it saying not supported and then need to make new profiles from scratch and can't just edit and save from existing.

### Describe alternatives you've considered

I can set them all to PLA but it opens up issues with mistakes and wrong material selected which then causes print issues etc

### Affected users and/or printers

everyone that uses this feature and PLA+ 
https://community.ultimaker.com/topic/33336-non-generic-material-selection-causes-profiles-to-show-not-supported/

### Additional information & file uploads

people on the forums keep complaining about this https://community.ultimaker.com/topic/33336-non-generic-material-selection-causes-profiles-to-show-not-supported/

fvrmr, 2021-10-14:
Hi @bodyboarder2528 thank you for your request.
When it shows not supported it means that there are no supported profiles for these materials. But you can still print with it. 
Also good to know there is a marketplace where you can download a lot of materials (But these materials are only for Ultimaker printers).
Or you could just edit the materials that are in the list, for PLA+ you have eSun. (This was also mentioned in the post you've linked)

Let me know what you think.

bodyboarder2528, 2021-10-14:
Also when you have alot of profiles, needing to remake them takes alot more time.

cyberjak2k, 2021-12-13:
I recently found finally that since 4.10 when all the eSUN pre-installed profiles did this ( not supported ) that the cause is the + in the name of the material type. even if you duplicate the material and then adjust as needed, it still says Not supported. 
BUT if you remove the + and put PLUS, or just leave PLA it works fine, with the predefined from the original material
So it's something about having that + in the material type name