Ultimaker/Cura #10623: Error in G-code when slicing a thin model

Eddy71, 2021-10-15:
### Application Version

4.11.0

### Platform

Windowd 7 64bit

### Printer

TT-1s

### Reproduction steps

I make slicing into layers in draft quality with the program's default settings.

### Actual results

As a result, I get a tear in the model, part of the model "hangs in the air" above the table surface

### Expected results

I want to get a normal slicing into layers. PrusaSlicer slices the same model normally.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[FG-Coil-Spule.zip](https://github.com/Ultimaker/Cura/files/7351451/FG-Coil-Spule.zip)

Eddy71, 2021-10-15:
Hi, GregValiant

> I think I'd be inclined to go with a .2 nozzle on that model AND use Arachne. The model is tall and weak. If you print with support then support removal is going to be tough.
> —

Thanks, but I sliced the model with PrusaSlicer and had already printed it quickly with acceptable quality with a .4 nozzle.

-- 
С уважением, Эдуард.
-----------------------------------
http://www.eddy.com.ua/

https://www.facebook.com/eddy.com.ua

Ghostkeeper, 2021-11-05:
> same settings.

I don't believe you! What is the value of the Skin Removal Width setting in Cura vs. PrusaSlicer? Are they the same? Well no, because it doesn't exist in PrusaSlicer. What about Outer Wall Jerk? Avoid distance? And similarly PrusaSlicer has a bunch of settings and features that Cura doesn't have. There is no realistic way to compare them other than through the g-code output.

And if there were, this is a print quality problem which I can't verify without having your printer. Even if I could guess where the real underlying problem lies that leads to this phenomenon, I could never verify that my changes fix the issue. Sorry :( To do something, we need to know what needs to change to the g-code output of Cura such that it prints better for you.

Eddy71, 2021-11-05:
Hi.

> I don't believe you! What is the value of the Skin Removal Width setting in Cura vs. PrusaSlicer? Are they the same? Well no, because it doesn't exist in PrusaSlicer. What about Outer Wall Jerk? Avoid distance? 

The comparison is obtained very simply: I took and installed Cura-beta and PrusaSliser. I loaded the stl model in both programs, put the size of my TT-1S print area (180x180x180mm), set the temperature for the PLA spool to 220 degrees (I picked it up earlier) and did not change any options. 
I'm just learning how to print and model 3D, so I try not to change unfamiliar options.

:)

-- 
С уважением, Эдуард.
-----------------------------------
http://www.eddy.com.ua/

https://www.facebook.com/eddy.com.ua