Ultimaker/Cura #10625: Unioned volumes in a single model don't print as a continuous layer

sbkivitz, 2021-10-15:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

The ball/head portion prints as a separate but touching object abutting to the neck/base.
This should print with continuous layers after the first few slices where there is an intentional gap between the two portions of the model.

Trying to print a strong neck piece for this model (looks like a chess pawn):
https://www.thingiverse.com/thing:3129886 

### Actual results

The top/ball head prints as a ball with inner/outer walls and the bottom neck/base print with their own inner/outer walls.

[image]

### Expected results

Perhaps it is an issue with the model - but I could not find a setting in Cura to allow me to print the model in a single continuous layer (towards the middle) for strength.

[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CE3PRO_neck.zip](https://github.com/Ultimaker/Cura/files/7354749/CE3PRO_neck.zip
[cura.log](https://github.com/Ultimaker/Cura/files/7354760/cura.log)
)

GregValiant, 2021-10-17:
Yes, it's the model.  The head and neck aren't really connected.  My guess is that whoever merged the models missed a trick.  What you had was Nearly Headless Nick from Harry Potter.  I fixed it.  There may be other problems.
[image]
[pawn.zip](https://github.com/Ultimaker/Cura/files/7361004/pawn.zip)

sbkivitz, 2021-10-19:
Awesome.  Thank you @GregValiant  & @fvrmr