Ultimaker/Cura #10630: No extrusion movement with 2nd extruder when printing with only that extruder

downerhouse, 2021-10-17:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Ender 5+, dual extruder (BMG x2), BTT GTR v1/M5 board

### Reproduction steps

1. Select print in Cura to print with 2nd extruder
2. Nothing selected to print in Cura with 1st extruder
3. Slice object and export gcode
4. Print

### Actual results

Print starts after bed / heater / G28 / G29 preparation.
Print head moves as if printing
No movement on 2nd extruder, so no filament deposited.

### Expected results

Cura should allow print objects to be printed with either extruder and print accordingly.
[PLA_X Test Print 2nd Extruder.zip](https://github.com/Ultimaker/Cura/files/7359942/PLA_X.Test.Print.2nd.Extruder.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

The attached file works fine if:
1. Printing test object with Extruder 1.
2. Duplicating the test object and assigning 1 object to extruder 1, and 1 to extruder 2.

Gcode attached

[PLA_04_Extruder 2_X Test Print.zip](https://github.com/Ultimaker/Cura/files/7359945/PLA_04_Extruder.2_X.Test.Print.zip)

downerhouse, 2021-11-01:
Small update.  I updated to Marlin v2.0.9.2 and the problem is the same.  I also bypassed Octoprint and tried to print directly from sdcard on the printer, just to remove another variable.  The printer still behaves the same.

I do not know what to change in the gcode to force different behaviour, so any advice welcome. 

I'm beginning to wonder whether the cause is due to the start / end gcode in the printer profile.  I'm using the stock code that gets generated by cura, but I changed it slightly for the dual extruder / ABL.  

This is what is currently in the start sequence:

START
------
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G28 ;Home
G29 ;ABL
M420 S1 Z2 ;Enable ABL using saved Mesh and Fade Height

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up

END
----
G91 ;Relative positioning
G1 E-2 F2700 ;Retract a bit
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positioning

G1 X0 Y{machine_depth} ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed

M84 X Y E ;Disable all steppers but Z

I can't find any information on the correct way to set the printer profile up for a combination of prints you may be making (could be left, right or both nozzles).  Any suggestions or input welcome.

Thanks

Ghostkeeper, 2021-11-01:
Is the feeder wheel moving at all? Perhaps the issue is priming?
Does your printer have a shared heater and/or a shared nozzle for both extruders, and is the Extruders Share Heater/Nozzle setting appropriately enabled/disabled?

downerhouse, 2021-11-01:
The BMGx2 extruder is a 2xdirect drive, 2x nozzle head, each with its own heater and thermistor assembly.  The 2nd extruder heats to the correct temperature, and everything on the printer moves as if it is printing - except that the extruder gears are not moving.  This only happens if I try to print only with the 2nd extruder.

As mentioned above, printing the item with either extruder 1, or extruder 1 & 2 works fine, using the same printer profile.

It is only when I try to print with the 2nd extruder alone does this issue occur.

GregValiant, 2021-11-01:
It sure looks like what you have there should work.
In regards to Ghostkeeper's suggestion, maybe this would make a difference.
...
M221 S100 ;Reset Flowrate
**T0**
G28 ;Home
G29 ;ABL
M420 S1 Z2 ;Enable ABL using saved Mesh and Fade Height
**G92 E0
T{initial_extruder_nr}** ;"initial_extruder_nr" must be in curly brackets
G92 E0 ;Reset Extruder
...
That will level with T0 active but switch to T1 for the purge lines (when necessary).  It might be worth a shot to see if it will extrude.  I do see a downside as T1 is initially heated and T0 is not so you would be leveling with a cold extruder which may have a booger on it.  If calling for T0 for leveling and then switching to T1 works and T1 extrudes properly then T0 can be heated for leveling and then turned off, if it's disabled and un-used for the current print.  I think.  Maybe.

Ghostkeeper, 2021-11-09:
I feel like we're kind of shooting in the dark here what could be the issue. Truth is that we can't find anything really wrong with the extruder switch g-code you posted, and the printers we can test with work fine. I'm not really sure what you can expect from Cura to resolve this.

We kind of need to know what we need to change in Cura's output g-code for your printer, otherwise this will not get fixed.

downerhouse, 2021-11-10:
Thanks guys. Sorry I have not got back to you, been hectic on other things. 

I'm going to devote some time tomorrow to have a go at hacking the g code to see what it will take to get it to work. 

I have to admit, I'm somewhat surprised that you have the same file printing successfully on your own printers. That points more towards a hardware or marlin issue that I have, but will do some more investigation. 

Will revert as quickly as I can.

downerhouse, 2021-11-11:
Hi, I changed the G code as suggested, and the behaviour is the same.  The bed / nozzle 2 heats up as per normal.  Then when it primes / tries to print, the head moves as expected, but the extruder does not move at all.

Ghostkeeper, 2021-11-18:
So what is it you would like us to do?

If the g-code clearly instructs the printer to move the extruder, but the printer doesn't move it, should we raise the issue with your printer manufacturer, Creality, or one of the add-ons you installed on it?

If you think the issue is still with Cura, what would you want us to do about it? Again, the g-code is perfectly fine as far as I can tell. What would you want us to change in the g-code output to make it work?

downerhouse, 2021-11-18:
I'll close the issue as we are getting nowhere. 

I don't have the knowledge of how to debug the g code to the level necessary to provide any other meaningful input. 

Other slicer applications work just fine with the same test. 

Thanks for your time.

closed 2021-11-18 by downerhouse

Ghostkeeper, 2021-11-25:
We'd have to wait and see if someone more knowledgeable about what the printer does can reproduce this problem as well, and see what is wrong with the g-code that would make it do that then.