Ultimaker/Cura #10638: Two Bugs

jonwienke, 2021-10-19:
### Application Version

4.11

### Platform

Windows 10

### Printer

Creality Ender 5 Plus

### Reproduction steps

I'm having issues loading files and slicing.

The first and most common is a conflict with serial ports. If I have any standard serial devices connected, Cura hangs, sometimes for hours, before opening files or slicing. With serial devices plugged in, file loads hang, but as soon as I unplug the serial device, files load immediately as normal.

In addition, When Cura is running, it also conflicts with other devices that use serial ports. For example, I have issues running ProScan, which communicates with scanners via COM ports, or the U-Blox GPS test/diagnostic utility, when Cura i s running because it's accessing and locking serial ports, and I can't get it to stop.

On top of that, I'm getting this error intermittently when slicing:
[image]

### Actual results

The file doesn't load until I disconnect all serial devices, or Cura doesn't slice. Here's my log file:
[cura.log](https://github.com/Ultimaker/Cura/files/7369917/cura.log)

### Expected results

The file should load and slice regardless of whether I have serial devices connected to the computer.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

I think I covered everything already.

jonwienke, 2021-10-19:
> Hi @jonwienke <https://github.com/jonwienke>  thank you for your report.
> About the serial ports, you can disable the plugin USB printing. Read here how to disable that: https://support.ultimaker.com/hc/en-us/articles/4407742254610-Ultimaker-Cura-interacts-with-microcontrollers
> About the failed slicing I would like to have a project file, to reproduce your issue.
>
> For next time please report separate issues for separate bugs, that makes it easier for us to keep track of bugs.

A project file is attached.

 

Turning off the USB printing plugin seems to have resolved the other issues.

fieldOfView, 2021-10-19:
Email attachments are filtered out by github. Please attach the log via https://github.com/Ultimaker/Cura/issues/10638 instead.

closed 2021-11-04 by no-response[bot]