Ultimaker/Cura #10703: Error: 'NoneType' has no attribute 'chop' -- saving empty project (I had setup TPU draft profile for old printer with no loaded model)

tyeth, 2021-10-29:
### Application Version

4.12.0-beta

### Platform

Windows 11

### Printer

Custom FFF Printer

### Reproduction steps

Select printer after loading cura, select profile for generic TPU.
Change initial printing temp to 225.
Skirt not brim, 4lines.
Save as tpu duplicate of draft profile.
40pc infil, 5layer top/bottom/shell.
Flow 95% general printing flow
Update custom draft tpu profile
Save empty project.

### Actual results

Error appeared saying something similar to  "Error: 'NoneType' object has no attribute 'chop'"

### Expected results

Successful project save message

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[cura.log](https://github.com/Ultimaker/Cura/files/7444260/cura.log)
line 3286

Can't save project so can't save project file, tried twice.

Included cura settings, saved using plugin "Profil Analyser" (yes no 'e')
[TPU Cura Settings.pdf](https://github.com/Ultimaker/Cura/files/7444271/TPU.Cura.Settings.pdf)

tyeth, 2021-10-29:
Since adding some objects and a few more settings adjustments it saves fine. Project and appended version of log attached
[cura.log](https://github.com/Ultimaker/Cura/files/7444523/cura.log)
[CFFFP_Snapmaker_small_laserbed_clip_v01.zip](https://github.com/Ultimaker/Cura/files/7444526/CFFFP_Snapmaker_small_laserbed_clip_v01.zip)

tyeth, 2021-10-29:
Why would someone want to save a project with no model loaded, well that's what I'm referring to as a template.
 I wish to save a template (cura project with no model yet) for certain profile settings that I can easily load a model into in the future.

fvrmr, 2021-11-01:
Hi @tyeth thank you for your report. 
Looking at your log the Smart Slice plugin is causing this. I tried to reproduce your issue and if I have the Smart Slice plugin installed it gives indeed this error. Without the plugin it doesn't give me the error when saving an empty project. 
Could you saving an empty project without the Smart Slice plugin installed?

tyeth, 2021-11-02:
Legend, thanks, I'll kill that off. Totally forgotten it was there which
probably ruins a few of my previous print tests 🙃

closed 2021-11-02 by tyeth