Ultimaker/Cura #10704: Slower speed for soluble materials

Doccos, 2021-10-30:
### Is your feature request related to a problem?

no

### Describe the solution you'd like

When i print ABS & BVOH both materials are not sticking perfectly together.
It would by nice to slow down the first( or more ) layer of an support interface and also to slow down the first(or more) ABS layer placed on the support interface.

### Describe alternatives you've considered

setting lower speed (increases print time dramatically

### Affected users and/or printers

Printers with two extruders

fvrmr, 2021-11-02:
Hi @Doccos thank you for your feature request. 
This is related to this issue: https://github.com/Ultimaker/Cura/issues/6656 but there was no decision made.

So I will bring this up with the team. Keep you posted!

Doccos, 2022-05-28:
Is there a plan to bring this feature into Cura5 ?
It looks the Cura4 version in the issue #6656 don't get a update to version 5