Ultimaker/Cura #10705: Show layer dependant estimated time

krikke-21, 2021-10-30:
### Is your feature request related to a problem?

Currently Cura shows the overall estimated print time.

### Describe the solution you'd like

After slicing, adjust and show the print time dependant on the layer currently showing on screen when a user uses the slider to previews/select layers. This layer dependant print time can be shown under the overall print time.

This allows user to see how much time the print will take before finishing by setting the slider to the current estimated layer the printer is currently printing. (This is a manual action)

### Describe alternatives you've considered

Check estimated print time on printer, but that's notoriously inaccurate

### Affected users and/or printers

All users

DoctorArr, 2021-10-30:
This sounds like an issue that needs to be fixed in the printer's firmware rather than in a slicer. I know everyone's workflow, current skill level, and budget play a part in the issue you are wanting to be fixed.

Have you tried (without knowing you're set up):

1) Asking the firmware maintainer to improve the accuracy?
2) Using Octoprint and its timer plugins, which also report back to the printer's display?

If you need help with point 2 I can point you in the correct direction.

fvrmr, 2021-11-01:
Hi @krikke-21 thank you for your feature request.
Like @DoctorArr mentioned the time estimation is related to firmware. We use the settings we get from the firmware and based on that we make a time estimation. 

You could also use the `Display progress on LCD` processing script. But that is related to your firmware.

Ghostkeeper, 2021-11-08:
The "alternative" in the issue template is indeed about checking time on the printer, which is sometimes inaccurate depending on the printer firmware. However Krikke-21's main feature request is not firmware dependent.

Effectively this is a duplicate of https://github.com/Ultimaker/Cura/issues/7683.

closed 2021-11-08 by Ghostkeeper