Ultimaker/Cura #10706: MacOS Monterey: Cura is not able to make multiple copys of one object

WolfgangDomroese, 2021-10-30:
### Application Version

4.11.0

### Platform

MacOS 12.0.1

### Printer

Ender 3V2

### Reproduction steps

Put one Object to Cura by Fusion360. Select the object and select "Ausgewählte vervielfachen 3".

### Actual results

Cura stops showing a message:
[image]

The message does not disappear. And no

### Expected results

Cura should place 3 additional objects. Id did so before I upgraded to Monterey.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No further comment.

liquiduk, 2021-10-30:
Same exact issue for me on Monterey

Imerch, 2021-10-30:
Аналогичная проблема!

fvrmr, 2021-11-01:
Hi @WolfgangDomroese thank you for your bug report.
This is a duplicate of #10249
I will close this issue since it is a duplicate.

closed 2021-11-01 by fvrmr