Ultimaker/Cura #10707: Cannot duplicate an object

liquiduk, 2021-10-30:
### Application Version

4.12.0

### Platform

OSX Monterey 12.0.1

### Printer

CR10 Smart

### Reproduction steps

Tried it on 4.12 and 4.11 and everything else seems to work but if I try and duplicate any stl on the buildplate it just shows the progress bar for duplication and gets stuck there/ never duplicating.

### Actual results

Just gets stuck

### Expected results

Duplicated object appear on buildplate

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Nothing to add.

fredlex3dprints, 2021-10-31:
I have exactly the same problem. 
Cura 4.11
OSX Monterey 12.0.1 
CR10V3

for now I just load the model more times instead of using multiplier....
But nog when you me changes you need to do it X times. not practical .

Xavier-BZT, 2021-10-31:
Cura 4.11 and Cura 4.12 beta
macOS Monterey 12.0.1

Same Issue, multiplication of models doesn't work

amuckin, 2021-11-01:
Same issue...
MacOS Monteret 12.0.1
Cura 4.11.0
Ender 3 v2

Manifested AFTER upgrade to Monterey.  duplicate object was working before upgrade.

fvrmr, 2021-11-01:
Hi @liquiduk thank you for your bug report.
This is a duplicate of #10249
I will close this issue since it is a duplicate.

closed 2021-11-01 by fvrmr