Ultimaker/Cura #10708: Make Overhang Printable makes vertical walls oscillate

BagelOrb, 2021-10-30:
### Application Version

4.11, 4.12 beta, more?

### Platform

Windows 10

### Printer

Ultimaker S5, probably any

### Reproduction steps

1. Load a model which has vertical walls
2. enable make overhangs printable
3. set angle to 70*

### Actual results

[image]

 Walls alternate in position. Odd layers have a small inset inward.

### Expected results

Straight walls
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[UMS5_proto.3mf.renamed.zip](https://github.com/Ultimaker/Cura/files/7446586/UMS5_proto.3mf.renamed.zip)

BagelOrb, 2021-10-30:
If I leave the overhang angle at 50* then the odd layers aren't generated at all:
[image]

fvrmr, 2021-11-01:
Hi @BagelOrb thank you for your report. 
Your model displays at the top that it needs overhang. When you fix your model with mesh tool, it doesn't happen anymore. 
Can you check if that is also your case?

BagelOrb, 2021-11-01:
Yeah. That's super weird. I never imagined that something like that could influence this feature, because after it's sliced all geometry should be the same regardless of normals.

Closing this issue because it's caused by a 'bad' model.

closed 2021-11-01 by BagelOrb