Ultimaker/Cura #10710: Configuración Kingroon KP3S

Joaquin1963, 2021-10-31:
### Application Version

4.12.0

### Platform

Windows 11 pro

### Printer

Kingroon Kp3s

### Reproduction steps

Desinstalar la impresora y volverla a  #instalar.

### Actual results

[image]

### Expected results

Ninguno.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fvrmr, 2021-11-01:
Hi @Joaquin1963 thank you for your report.
You can delete your configuration folder: `C:\Users\<your username>\AppData\Roaming\cura\<Cura version>`
This will delete all your settings and profiles. 
Or you can only delete the extruder configs 
`C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\extruders\<your extruder files.cfg>`

Let me know if that works.

Ghostkeeper, 2021-11-08:
Deleting a printer shouldn't cause corrupt configurations though. Perhaps Joaquin had multiple instances of Cura open at the time of removal? That could cause it.

I'm unable to reproduce this bug otherwise, so far.

closed 2021-11-15 by no-response[bot]