Ultimaker/Cura #10711: Tree support footing placement poorly impemented

mfratto, 2021-10-31:
### Application Version

4.11.0

### Platform

Windows 10 Pro 21H1 19043.1237

### Printer

Creality CR-10V3

### Reproduction steps

The bug is that support placement is making what seems to be the wrong choice on where to place support.

In the attached zip is the model STL, a screen shot, and the Cure project file.
1. I added the STL to the workspace, selected my profile, and sliced.
2. The result is that the support foot is placed inside the model when there is clearly space to put it on the plate.
3. If I select "Support Placement-> Touching Buildplate", then support is not added. 
I had to add a support blocker because the 
[Desktop.zip](https://github.com/Ultimaker/Cura/files/7449111/Desktop.zip)
e 

### Actual results

[image]

The support (light blue) is placed inside the box.

### Expected results

The support should be placed on the buildplate.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Everything is attached. Thank you

fvrmr, 2021-11-01:
Hi @mfratto thank you for your report. 
I will bring this up for discussion with the team. Keep you posted!

fvrmr, 2021-11-01:
Also I believe that your model doesn't need support there. You could use bridging settings, since your overhang is supported on both sides of your model.

fvrmr, 2021-11-02:
I have discussed it with the team. And we decided to defer this bug, since it is probably model specific. So it is not a priority for us to fix this. I hope you understand.

Ghostkeeper, 2021-11-08:
Perhaps the widening of the branch made it suddenly not fit any more inside the hole in that model. It should account for that, but I've seen that in rare cases the MoveOutside it does to avoid objects can still end up inside an opposite piece. From that moment on it may think there is no way out to the build plate any more.

DerGenaue, 2021-11-26:
If you want, you can try out this alternative Tree Support implementation that a friend of mine is working on, maybe that one works a better for you:
https://github.com/ThomasRahm/CuraEngine

closed 2023-06-09 by github-actions[bot]

DerGenaue, 2023-07-04:
FYI: This issue probably has been solved in Cura 5.4