Ultimaker/Cura #10714: Sliceing failed

FraSa67, 2021-11-01:
### Application Version

4.11.0

### Platform

MacOS 11.6 M1

### Printer

Artillery Sidewinder X1

### Reproduction steps

Loaded a model*, slicing started, after about 50% progress got a message that slicing failed and I should open a bug report (I can provide the Modell and I will contact the model creator on this issue as well)

### Actual results

the slicing process seems to be stuck in a loop

### Expected results

file ready to save on drive  for printing

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Church-Back-Roof-15mm-v1.7.stl.zip](https://github.com/Ultimaker/Cura/files/7452636/Church-Back-Roof-15mm-v1.7.stl.zip)

fvrmr, 2021-11-01:
Hi @FraSa67 thank you for your bug report.
Could you also share a project file (file - save project)
With only your stl I could not reproduce your issue:
[image]

FraSa67, 2021-11-01:
it seems this problem occurs only when using the "dynamic 0.16mm" profile. using the 0.12mm profile the slicing was successful. So no big deal on my behalf, you might close the issue if you like. Thanks a bunch for great software!
[ASX1_Church-Back-Roof-15mm-v1.7.gcode.zip](https://github.com/Ultimaker/Cura/files/7453343/ASX1_Church-Back-Roof-15mm-v1.7.gcode.zip)

UxuginPython, 2021-11-02:
I think I might have the same issue (#10720)

fvrmr, 2021-11-04:
Hi @FraSa67 I sliced your stl on dynamic quality with an internal build of 4.12. And there I can slice your model. So when the 4.12 is out, you can slice this! 
The 4.12 beta didn't work, so you will have to wait for the 4.12 stable release.

[image]

I will close this issue, since it will be fixed in 4.12!