Ultimaker/Cura #10715: Effect on extruder1 when disabling extruder2

helgiHafst, 2021-11-01:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Ultimaker 3

### Reproduction steps

1. I am using a 0.8AA in extruder 1 and I am using a 0.4AA in extruder 2 
2. I disabled extruder 2
3. Then started the print with extruder 1 and the initial layer does not stick to the print bed as it should.
4. I enabled the extruder 2 again 
5. Now the initial layer sticks to the bed just fine. 
6. I can see that the Initial layer height for extruder 1 jumped from 0.27 to 0.53 when I disable extruder 2.  

### Actual results

If I don´t change the initial layer height to 0.27 on extruder 1 after disabling extruder 2 the initial layer will not stick to the print bed as it should

### Expected results

The Initial layer height should stay the same on extruder 1 even though I disable extruder 2.  

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fvrmr, 2021-11-01:
Hi @helgiHafst thank your report. 
This is expected behaviour based what is in our code. When the nozzle is disabled it picks the minimum value based on the profiles, which is 0.53m. But I can understand that this is not desired behaviour and will fail your print. 
I will discus this internally. Keep you posted.

Ghostkeeper, 2021-11-09:
This effect is essentially why we ended up adding the functionality to be able to disable an extruder. The expectation is that disabling the extruders you're not using results in a better print, because it's then able to use the settings exactly as specified in the profiles for the extruder you are using.

The Initial Layer Height is specified to be [2/3rds of the nozzle size](https://github.com/Ultimaker/Cura/blob/73ca50d958d08c5127fff97f3e75f414ca5cf2c4/resources/definitions/ultimaker3.def.json#L121) in the Ultimaker 3 profile. When there are multiple extruders involved where this would lead to a different value, Cura [picks the lowest of them](https://github.com/Ultimaker/Cura/blob/73ca50d958d08c5127fff97f3e75f414ca5cf2c4/resources/definitions/fdmprinter.def.json#L792) for this setting. So if the 0.4mm nozzle is active, it's expected to end up at 0.27mm (2/3rds of 0.4). If only 0.8mm nozzles are active, it's expected to end up at 0.53mm (2/3rds of 0.8mm).

So the real issue here is that the default values for the 0.8mm profile don't seem to work well for HelgiHafst. Apparently lowering the Initial Layer Height would make it more reliable. Perhaps @pkuiper-ultimaker would like to hear this feedback?

pkuiper-ultimaker, 2021-11-09:
Interesting. A initial layer height of 0.53mm should be enough for PLA to stick using a 0.8mm nozzle and 0.2mm layer height. We might need to lower the initial layer height  in the print profiles if it reproduces. I will try to reproduce it on my own machine and will ask our adhesion expert about this case. 

Note: Adhesion is very depend on the state of the build plate. Make sure it is clean and apply glue and a brim when in doubt.

pkuiper-ultimaker, 2021-11-09:
[image]

pkuiper-ultimaker, 2021-11-09:
I just printed a block on my S5 (with brim) with silver PLA 0.8mm print core and 0.2mm default profile and with an overwrite on the initial layer height of 0.53mm.

The adhesion was very good. I had a lot trouble getting it loose.