Ultimaker/Cura #10716: Weird Fonts in Settings

michaelgrossklos, 2021-11-01:
### Application Version

4.11

### Platform

Big Sur (M1)

### Printer

Elegoo Neptune 2

### Reproduction steps

1. Open Settings ([https://imgur.com/a/2dgSSz8](https://imgur.com/a/2dgSSz8))

### Actual results

Weird looking fonts in dropdown fields

### Expected results

Correct fonts

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

fvrmr, 2021-11-01:
Hi @michaelgrossklos thank you for your report. 
This is a duplicate of https://github.com/Ultimaker/Cura/issues/8993
I will close this issue since it is a duplicate.

closed 2021-11-01 by fvrmr

julianCast, 2022-08-09:
This is happening to me, but **without M1** chip model, only happening on Big Sur:

Application Version
4.11

Platform
Big Sur (Intel)