Ultimaker/Cura #10717: Cura will not ensure objects are separate and will not multiple/arrange

xizahn, 2021-11-01:
### Application Version

4.11.0

### Platform

MacOS 12.0.1

### Printer

Anycubic I3 Mega Pro

### Reproduction steps

1) When importing multiple STL files, Cura will not ensure they are separate despite the setting being check marked.
2) When right clicking on a selected model and selecting multiply, Cura will not multiply the object, it will open the window that it is working, but the progress bar never moves.
3) when right clicking on a selected model and selecting arrange models, Cura does not arrange the models on the build plate.  nothing happens.

### Actual results

Cura failed to perform the 3 tasks above.

### Expected results

Cura should have either kept the models apart or multiplied them or arranged them.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

This only started when the iMac updated to Monterey v12.0.1

I have tried deleting Cura, removing the folder from /library/application support, restarting and reinstalling -- same results
I have also tried installing 4.12 and it too has the same issues.

[cura.log](https://github.com/Ultimaker/Cura/files/7454752/cura.log)

fvrmr, 2021-11-02:
Hi @xizahn thank you for your report. 
1. Could you share a screenshot of what is happening?
2. This is a known bug, see here: https://github.com/Ultimaker/Cura/issues/10249
3. This is also a known bug, see https://github.com/Ultimaker/Cura/issues/10658 which is fixed for the stable release of 4.12

For next time please report everything separately, so it is easier for us to keep track of all the bugs. :)

xizahn, 2021-11-03:
Thanks, sorry, my first GitHub bug report... So, below I have a series of screenshots -- the first shows my Cura settings with Ensure models are kept apart checked
[image]

Second photo shows importing the first model ---
[image]

Third photo shows what happens when I import subsequent models --
[image]

Last photo reiterates the third photo, models are not kept apart --
[image]

fvrmr, 2021-11-04:
Thank you for the screenshots. I can't reproduce this on Windows. 
I need to check this with a colleague who is using Mac to see if we can reproduce this as well. 
I will get back to you when I have more information!

fvrmr, 2021-11-22:
Closing this issue since this is solved in 4.12.1 which we released last week.
This bug was related to https://github.com/Ultimaker/Cura/issues/10249

closed 2021-11-22 by fvrmr