Ultimaker/Cura #10718: German translation for estimated time

fralut, 2021-11-01:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Ultimaker S3

### Reproduction steps

When i select the "monitoring" tab, the estimated time with german language is a mixture of german an englisch. e.g.

Finishes morgen at 11:08

### Actual results

Correct would be:

Fertiggestellt morgen um 11:08

### Expected results

Therefore the word "Finishes" needs to be replaced with "Fertigestellt" and "at" with "um"

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fvrmr, 2021-11-02:
Hi @fralut thank you for your report. 
I will bring this up with the team!

Ghostkeeper, 2021-11-08:
Good find! This was a small oversight. Fixed now.

closed 2021-11-08 by Ghostkeeper

fralut, 2021-11-09:
Unfortunately, the change was not implemented as i recommended. The new translation "auf" instead of "um" is a bad german translation and better would be really my recommendation.

[image]

commit db637802 (2021-11-08): First translate, then process replacement keys

If we first process the replacement keys, the replaced string can't be found in the translation database, so you don't get a translation of the phrase 'Finishes x at y'.

Fixes #10718.