Ultimaker/Cura #10719: Random Zero Added towards the end of the outputted Gcode.

PriddyPrints, 2021-11-01:
### Application Version

4.12.0 BETA

### Platform

Mac OS Monterey 12.0.1

### Printer

Craftbot XL

### Reproduction steps

Anytime a file is sliced with my current setup, a random zero is added at the end of the Gcode, which the machine cannot read and throws an error while preemptively checking the code. No such value is present in the machine settings end of Gcode.

### Actual results

Zero added to the end of a Gcode. It is the only value on that line. 

### Expected results

Assuming the zero should not be present. Not sure what it could be tied to. 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Attached are screenshots to help show its location, 3MF, outputted Gcode, and log files. If anything else is needed, please let me know.

[image]
[image]
[Archive.zip](https://github.com/Ultimaker/Cura/files/7456434/Archive.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/7456435/cura.log)

fvrmr, 2021-11-02:
Hi @PriddyPrints thank you for your report. 
Could you share your project file (file - save project)? The 3mf file you've shared is not a project file, I know it's a bit confusing.

PriddyPrints, 2021-11-02:
Hi, @fvrmr,

That's odd. I am able to reload the file on my end as a project, rather than just an exported 3MF. I've gone ahead and resaved it, though. If you run into any issues with this copy, let me know.

[Cura Infinity.3mf.zip](https://github.com/Ultimaker/Cura/files/7462093/Cura.Infinity.3mf.zip)
[image]

GregValiant, 2021-11-02:
I can't open it as a project either.

However...this is with 4.12beta and I just noticed that I get the same thing with my Ender 3 Pro.
@fvrmr I don't think you'll need a project file.  All my slices show this since I started with 4.12 but the Ender just moves on.

;TIME_ELAPSED:15063.312858
G1 F2100 E1682.70117
0
M204 S500
M107
;Start of End GCODE
G91 ;Relative positioning

PriddyPrints, 2021-11-02:
I appreciate it, Greg! Not sure why my 3MF is not behaving properly.

GregValiant, 2021-11-02:
I'm not sure why it doesn't open as a project (it does pull the model out though).  It doesn't appear that there is a definition file for your printer in my installation of 4.12.  If you used a definition file you received from someplace then that might be it.  If you used the generic fdm printer definition and altered it to suit then the file should have opened as a project.
I get the same not-so-random "0" but as I say, the Ender ignores it and moves on.  I've only been using 4.12 for a few days and hadn't looked at a gcode file in that time.  The "0" is in all the gcodes I generated with 4.12 though.
Looking at it now I think it might be intended to be M104 S0 as that line no longer appears in my gcode files until the very end and it used to appear in both places.

PriddyPrints, 2021-11-02:
You hit the nail on the head. The definition file was sourced elsewhere and loaded in, rather than modified from the generic premade. At the very least, I'm glad to see that it only looks to be 4.12 related and not profile specific.

GregValiant, 2021-11-02:
Since your printer hates that zero I think you will need to use 4.11 or go into each gcode file and take the zero out (or make it M104 S0).
**This is from a file sliced with 4.11**
;TIME_ELAPSED:2209.195114
G1 F2100 E2376.97882
M140 S0
M205 X30 Y30
M107
;Start of End GCODE
G91 ;Relative positioning
**This is from the same model sliced with 4.12**
;TIME_ELAPSED:2161.959830
G1 F2100 E2704.96301
0
M205 X30 Y30
M107
;Start of End GCODE
G91 ;Relative positioning

PriddyPrints, 2021-11-03:
Thanks for confirming that! I'll downgrade to 4.11 or look into a post processing script in the meantime.

GregValiant, 2021-11-03:
That's my bad.  It was **M140** S0.  Without it - the bed doesn't turn off unless there is one in the End G-Code.

fvrmr, 2021-11-05:
Thank you @GregValiant 
I could reproduce this also on Ultimaker printers, witch also ignores the 0
So I have discussed this with the team and I created a ticket on our backlog.

Devs see CURA-8673

nallath, 2021-11-05:
The reason it can't open the project file is because it's a custom machine that was added by hand. Other cura versions don't have that machine.