Ultimaker/Cura #10725: Can multiply object in Mac OS 12.0.1

izzycura, 2021-11-02:
### Application Version

4.10, 4.11, 4.12 beta

### Platform

MAC OS Monterey 12.0.1

### Printer

JGAurora

### Reproduction steps

1. Right Click on object
2. Multiply Selected
3. Notification window with time bar comes out.

### Actual results

Nothing happens window bar stays open and it just hangs.

Repeat and another window bar opens and continues to hang

### Expected results

It should multiply my object. I also tried different objects.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

nallath, 2021-11-03:
Duplicate of #10249

nallath, 2021-11-03:
For the next time; Please fill in the entire template (eg; include log files). We need that information to help diagnose the issue and provide a fix.

closed 2021-11-03 by nallath

tommyhlc, 2021-11-04:
I have the same issue, thought the BETA ver. can help, but still not working...