Ultimaker/Cura #10726: Stringing and weird height issues, also same project file generates _very_different Gcode

mtempels, 2021-11-02:
### Application Version

4.11.0 vs 4.12.0Beta

### Platform

OSX 12.0.1

### Printer

Ender 5 Pro / Marlin

### Reproduction steps

Open 3mf file in 4.11 vs 4.12, slice, compare Gcode
the 4.11 file prints flawless, no stringing, no problems
the 4.12 file has a ton of stringing and the nozzle seems to hit the infills while moving around

### Actual results

Different gcode from same file, with a worse result from the 4.12 slice

### Expected results

Similar gcode and same quality prints

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Archief.zip](https://github.com/Ultimaker/Cura/files/7463403/Archief.zip)
This is the file with the project and 2 resulting gcode files

fvrmr, 2021-11-04:
Hi @mtempels thank you for your bug report.
If I open your gcode in Cura, I see that in 4.11 and 4.12 beta this: 
[image]

[image]

Did you print both of files? Cause looking at the g-code preview it looks the same.
Does the prints looks the same as above?

When I reset your profile settings and use the standard quality 2.0 I see this: 
[image]

About print quality it is always hard for us to help with what is causing this, since we don't test on third-party printers.

Ghostkeeper, 2021-11-09:
Cura 4.12 is expected to be different from Cura 4.11.

In your case, the main difference appears to be a bug fix where it was overextruding infill in the same areas as gap filling. This is your project in Cura 4.11:

[image]

And this is the same part of the same project in Cura 4.12:

[image]

So in 4.12 the slice no longer contains overextrusion of infill in gap filling areas. As a result, the printing time and material usage is significantly reduced.

I would blame the stringing issue on the usage of Arc Welder. But in any case, to fix stringing we'd need to know what we can do to improve it for you. We can't test with your printer, so we can't make any change to fix it; we wouldn't know if it fixed it.

Ghostkeeper, 2021-11-11:
I'd suggest to try to analyse the problem with the pointers given in the Settings Guide plug-in: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/stringing.md

I'm not going to try to analyse a stringing problem remotely via pictures and guessed setting improvements. If you know of something we can improve, please tell! But otherwise we'll close this issue, since it's expected that the g-code output is indeed different in 4.12.

closed 2021-11-11 by Ghostkeeper

mtempels, 2021-11-11:
I just updated to 4.12.0 final, reset all parameters to my printer and the 0.20mm profile and printed a perfect part..

I'm going to try this project once more and reset all parameters to default.. 

I'll keep you posted!