Ultimaker/Cura #10730: Outer wall uniformity using [Minimum layer time]

Akujiproxy, 2021-11-03:
### Is your feature request related to a problem?

When setting a [minimum layer time], the print speed of the outer walls can vary drastically on certain layers of a print. This variation is visible in the end result. The print is still smooth but there seems to be a difference in the way it reflects light.

[image]

### Describe the solution you'd like

I believe a good solution to this issue is to have Cura determine what the slowest outer wall speed would be throughout all the layers of the print when a minimum layer time is enabled and have an option to apply this speed to every layer. By equalizing the outer wall speed, the uniformity should improve. 

### Describe alternatives you've considered

An alternative would be to set a slower outer wall speed but that would increase print time. The suggested solution would make it possible to decrease print time overall.

### Affected users and/or printers

This could benefit anyone that uses the [minimum layer time] feature (printing at higher speeds).

fvrmr, 2021-11-05:
Hi @Akujiproxy thank you for your feature request.
Could you share your project file with me of the image you've shared? 
Is this only relevant for this model or do you have any other models where you noticed this as well?

Akujiproxy, 2021-11-05:
@fvrmr Unfortunately I'm out of PLA and in the midst of tuning my printer for PETG. I also don't have the file anymore. But I presume this issue is on every single print. The issue is with the speed difference in the outer walls when using a minimum layer time. when a layer is faster than minimum layer time, the outer wall is slowed down. And if the speed is a determining factor in how the filament dries up. So if speed variates, so does the glossyness of the layers. Once I've dialed in PETG, I'm going to be printing this model again and can provide you with it if you want.

Ghostkeeper, 2021-11-10:
This is indeed a pretty well known issue causing banding/striping. A complicating factor on your print however is also that this is where the model splits into two parts for the ears of Calicat. This causes travels and retractions which mess with the flow too, causing similar effects.

So the suggestion here is to apply the minimum layer time to every layer, at least with printing the outer walls.

This could be a bit of a problem architecturally because the layers are calculated in parallel, and this minimum layer time is one of the steps halfway through calculating the layer (well, a bit towards the end but not the last step). This would require all of the layers to complete calculating with a barrier, then determine the slowest one, then continue with re-scaling the speed and the rest (e.g. time estimates) in parallel again. It's possible, but requires a bit of re-structuring.

We currently downscale all of the speeds linearly to reach the Minimum Layer Time. This can downscale down until the slowest line hits the Minimum Speed. So for instance, if the Minimum Speed is 10mm/s, the Outer Wall Speed 20mm/s and the Inner Wall Speed 30mm/s, the speed of that layer could be reduced by up to 50%, resulting in an Outer Wall Speed of 10mm/s and an Inner Wall Speed of 15mm/s. If the Outer Wall Speed is scaled down separately from the rest of the layer, this would also need to get a re-think. It could be that on a particular layer you'd end up at a speed scale of 75% to achieve the Minimum Layer Time, but because the Outer Wall Speed needs to be reduced further for different layers and so the print is slower, that would be too much of a downscale and 85% could already be sufficient. This then ends up with some circular logic because the Outer Wall Speed itself then depends on that 85% as well. In theory this circle is broken by the slowest layer where this doesn't apply, but this needs to be cascaded properly.

And one last thing that could become a concern for print quality: We currently scale the entire layer linearly. If Outer Wall Speed prints at 50% speed, then the Inner Wall Speed prints at 50% speed too. That way the profiles can still try to maintain stable relative flow rates, for instance by making the inner wall print faster but with reduced line width such that the flow between the two walls is the same, improving dimensional accuracy by preventing flow changes. Changing the outer wall alone breaks this linearity. And it gives the profile author no control because the minimum layer time induced speed is determined by the shape of the model.

So while I think it would be useful to try to reduce (but not eliminate) banding, I also think there are some hurdles that must be thought out well and some print quality issues that would need to be taken into account.

Akujiproxy, 2021-11-10:
@Ghostkeeper Perhaps my suggestion does not fit as a solution within the way Cura and its feature set is designed right now. But it seems you have a better understanding of its workings and thus may be able to come up with a better solution.

Ghostkeeper, 2021-11-12:
As with any of these reports, our job is to solve the problem, which may not always be the suggested improvement but hopefully something that (also) works :wink: 

We discussed this with the Cura team today and ended up deciding to bring this over with the process engineers of Ultimaker to ask them what they would see as a solution to this problem.