Ultimaker/Cura #10731: Cura crashed during slicing

PeterGillham1, 2021-11-03:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Creality CR-10S Pro V2

### Reproduction steps

Loaded STL file @ 100% - sliced fine ( 7.5 hrs estimated time)
Reduced size to 60% and on re-slicing Cura crashed
[CCR10SPRO_Merry_Christmas.zip](https://github.com/Ultimaker/Cura/files/7467959/CCR10SPRO_Merry_Christmas.zip)

### Actual results

When re-slicing at 60% Cura crashes

### Expected results

Re-slicing to 60% should be OK

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.zip](https://github.com/Ultimaker/Cura/files/7467987/cura.zip)

fvrmr, 2021-11-04:
Hi @PeterGillham1 thank you for your bug report.
I could reproduce this indeed in 4.11 but I could slice it in 4.12 beta. 

[image]

Let me know if this works for you as well!

PeterGillham1, 2021-11-04:
Thank you, yes works fine in 4.12 Beta so assume issue fixed in 4.12 :-)

closed 2021-11-04 by PeterGillham1