Ultimaker/Cura #10735: Multiply selected waits forever

TriviumGames, 2021-11-03:
### Application Version

4.11.0

### Platform

MacOs

### Printer

Creality CR-10, but I haven't sliced yet

### Reproduction steps

I load a simple model, select it, right click and say "multiple selected". I choose 3 copies.

### Actual results

A little dialog pops up that says "Placing Objects.   Multiplying and placing objects" with an empty progress bar.  The progress bar never fills, and the object never gets multipled

### Expected results

In previous versions this feature worked, and it did a nice job of laying them out even.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I attached an example file though it's done it for a lot of other stl files for me.  The file loads fine and slices and prints fine, and I can load multiple copies and move them myself if I want.

[ServoCoupler v11.stl.zip](https://github.com/Ultimaker/Cura/files/7470816/ServoCoupler.v11.stl.zip)

mrrstrat, 2021-11-04:
Application Version

     4.11.0

Platform

    MacOs/Monterey

Printer

Creality Ender 3 V2 - but the issue is in Cura

Reproduction steps

I load a simple model, select it, right click and say "multiple selected". I choose 9 copies.

Actual results

A little dialog pops up that says "Placing Objects. Multiplying and placing objects" with an empty progress bar. The progress bar never fills, and the object never gets multiplied (ME TOO!!)

Expected results:

In previous versions this feature worked, and it did a nice job of laying them out even. Loading objects manually one at a time also used to work. Now all of the objects are overlayed at the same spot.

Checklist of files to include:

Additional information & file uploads:

I can attest this issue: I have Cura 4.11, on a 2019 MBP 13 i5 running Monterey. Everything worked before the Monterey upgrade but the only issue for me is the multiply does not work at all and reloading the same object overlays it in the same spot requiring manual placement instead of it happening automatically before the upgrade.

TriviumGames, 2021-11-04:
Oh I upgraded Cura at the same time I upgraded MacOS (to Monterey) so I assumed it was a Cura problem not (somehow??) a MacOS problem.

mrrstrat, 2021-11-04:
Monterey needs some special accommodation that remains yet to be identified to make Cura work apparently. The problem is that it does not seem to be a security or access thing: why multiplying an object would not work is something the Cure dev team might be able to shed some light on.

fvrmr, 2021-11-04:
Hi @TriviumGames thank you for your bug report.
This is a duplicate of #10249
I will close this issue since it is a duplicate.

closed 2021-11-04 by fvrmr