Ultimaker/Cura #10736: Allow Bridge Skin Density > 100%

qwertymodo, 2021-11-04:
### Is your feature request related to a problem?

When printing a bridge, the lack of a previous layer to press the lines flat means that they are "taller" than usual, and therefore "narrower" leading to gaps between lines on the first (and sometimes second or more) layer of the bridge.  In order to counteract that, it would make sense to print the lines of a bridge closer together than normal.  There is a setting which affects the line spacing of bridge areas, called "Bridge Skin Density", but it is limited to only accept values <= 100%, and anything > 100% results in a slicing error.

### Describe the solution you'd like

It seems like this is just a hard-coded limitation that refuses to accept any value over 100% for this setting.  I think this restriction should be removed to allow bridges to be printed with lines closer together.

### Describe alternatives you've considered

There is another setting "Bridge Skin Flow" that can be set to > 100% to fill in these large gaps, but high flow in bridge lines causes other issues, like additional sagging, and slower cooling, so in a lot of cases, it would be better to print with reduced flow rate and increased density.

### Affected users and/or printers

Affects all printers/users when printing models that include bridge geometry.

smartavionics, 2021-11-04:
See https://github.com/smartavionics/Cura/releases for a cura variant that provides what you're asking for.

qwertymodo, 2021-11-04:
Thanks for that link.  It's still a feature I'd like to see upstreamed.

Ghostkeeper, 2021-11-11:
Yeah, our policy is to set the limits for settings to be what is physically impossible (e.g. slower than 0mm/s) or algorithmically impossible (e.g. so many gradual infill steps that the infill line distance gets integer overflowed). There is no physical or algorithmic limit to cap that to 100%. We should probably have corrected that with the PR that introduced the feature.

I've changed the limit into a warning now.

closed 2021-11-11 by Ghostkeeper

smartavionics, 2021-11-11:
Don't forget that the 2nd and 3rd bridge skin layers have their own skin density limits.

Ghostkeeper, 2021-11-18:
Ah yes, indeed. Thanks. I've expanded the fix to include those as well now.

commit 61711071 (2021-11-11): Remove upper limit of bridge skin density

There is no theoretical or algorithmic limit here. We may warn the user, but not prevent them from changing it.

Done as a five minute fix. Fixes CURA-8693. Fixes #10736.