Ultimaker/Cura #10743: Wont slice large files

xxxDrewedxxx, 2021-11-04:
### Application Version

4.11

### Platform

Windows 10

### Printer

Ender 3v2

### Reproduction steps

Attempted to slice a large file.  727mb.

### Actual results

Cura did not slice.  The progress bar never moved.  I let it run all night long and still nothing by morning

### Expected results

I expected it to slice.
[No Slice Litho.zip](https://github.com/Ultimaker/Cura/files/7477075/No.Slice.Litho.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/7477082/cura.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

This file worked when the size was reduced to under 350 mb

fvrmr, 2021-11-05:
Hi @xxxDrewedxxx thank you for your report.
Could you share a new project file (file - save project)? 
I can't open this project file.

Ghostkeeper, 2021-11-11:
The zip file contains a 0-byte empty file.

This issue is a duplicate of https://github.com/Ultimaker/Cura/issues/3105 . There is a 500MB limit to the size of the vertex buffer we can send over to CuraEngine, imposed by Protobuf. This is not necessarily the same as the file size, but some mesh formats which compress their contents, like 3MF, can get a lot of triangles in quite efficiently.

closed 2021-11-11 by Ghostkeeper