Ultimaker/Cura #10745: Unable to "Multiply" (duplicate) objects

KenLucke, 2021-11-04:
### Application Version

4.12 beta

### Platform

Mac OS 12.0.1 (Monterey)

### Printer

n/a

### Reproduction steps

1. Load object
2. Attempt to multiple object, either via control-click (contextual menu), object list, or main menu bar

### Actual results

Object does not multiply, regardless of method used or number of times to multiply.  Notification box shows up saying it's being multiplied and placed, but it never happens, AND the box never goes away or allows closure.  I can add additional objects by simply opening the model file again, but that defeats the purpose of having a "multiply" command, and is a royal PITA for a large number of objects, or doing multiple objects at one time.

I just updated to Monterey (MacOS 12.x) from Catalina (MacOS 10.15.7) yesterday, and did have this behavior before (multiplying an object always worked, even in 4.12 beta).  Not sure if that's affecting things or not.

This happens regardless of the model or project file I attempt.

### Expected results

Should multiply.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

I cleared the logfile, then restarted Cura to get a clean one.
[cura.log.zip](https://github.com/Ultimaker/Cura/files/7479345/cura.log.zip)

Screen recording of the issue, using a simple model, attempting to just duplicate it once
Uploading screencast 2021-11-04 15-46-25.mp4…

fieldOfView, 2021-11-05:
Duplicate of #10249

fvrmr, 2021-11-05:
Closing this since it is a duplicate.

closed 2021-11-05 by fvrmr

KenLucke, 2021-11-05:
Sorry, I didn't see the previous one.  My fault.