Ultimaker/Cura #10746: Cannot multiply objects

ukitconsultant, 2021-11-05:
### Application Version

4.12.0-BETA

### Platform

MacOS Monterey

### Printer

Creality Ender 3

### Reproduction steps

Added a STL file by dragging and dropping or by using the file browser. Right clicked on the filename and left clicked on multiply. Typed in "2", then either pressed return or clicked OK. 

### Actual results

A box appears stating that it is"Placing Objects" with the text "Multiplying and placing objects", with an empty progress bar and does not progress. 

After some time exiting the program and restarting it, then re-trying the process results in the same issue.

### Expected results

I had expected the object to be multiplied into 2 identical copies of the same object. 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7481567/cura.log
[image]
[image]
[image]
)

This bug also seems to effect 4.11.0, as I updated to the beta in order to see if that resolved the issue, but it did not.

ukitconsultant, 2021-11-05:
Just to add. I am on an 2017 Intel iMac Pro with an Intel 18-Core Xeon 2.3Ghz and 128GB of RAM, as well as the dual Radeon Pro Vega 64 (16GB of GDDR), so I do not think it's a hardware issue. It worked before I updated to Monterey, and appears to have been introduced as an issue, possibly because of some of the core OS changes in Monterey.

fieldOfView, 2021-11-05:
Duplicate of #10249

fvrmr, 2021-11-05:
Closing this issue since it is a duplicate.

closed 2021-11-05 by fvrmr