Ultimaker/Cura #10747: Bad Skirt on a print of multiple parts

Joachim-R, 2021-11-05:
### Application Version

4.12.0 Beta

### Platform

macOS Catalina

### Printer

Ender 3v2

### Reproduction steps

using https://www.thingiverse.com/thing:3796670:

Slicing the https://cdn.thingiverse.com/assets/da/23/14/40/38/Slotter_1.stl leads to a wrong skirk around only one of the 5 dishes.
[image]

### Actual results

See image

### Expected results

Skirt around all dishes

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

-

fvrmr, 2021-11-05:
Hi @Joachim-R thank you for your report.
Looking at your model in Blender, you can see that the five sticking out 
[image]
So Cura see this as the first layer. If you change your profile to dynamic quality it will print the skirt correctly.

See also https://github.com/Ultimaker/Cura/issues/9420

I will close this issue since it is a duplicate. But please let me know if you have more questions.

closed 2021-11-05 by fvrmr

Joachim-R, 2021-11-05:
Just a note:
You/Me must not print this stuff using "Super Quality". That is all.
Thanks for the hint.