Ultimaker/Cura #10749: Models are not kept apart when opening multiple

deadlyknapsack, 2021-11-05:
### Application Version

4.11.0

### Platform

Mac OS

### Printer

Ender 3 V2

### Reproduction steps

Opening multiple files while "Ensure models are kept apart" is checked, will let all models overlap in center of the buildplate. Have to manually drag them apart. In previous versions, this worked fine.

### Actual results

Files/models are on top of each other.

### Expected results

Distribute models on build plate, to ensure they are kept apart.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]

fieldOfView, 2021-11-07:
Same underlying issue as #10133 and #10249

fvrmr, 2021-11-08:
Hi @deadlyknapsack thank you for your bug report. 
This is a duplicate of https://github.com/Ultimaker/Cura/issues/10717 I will close this issue since this is a duplicate.

closed 2021-11-08 by fvrmr