Ultimaker/Cura #10750: Cura not opening 3mf files

userosos, 2021-11-05:
### Application Version

4.11.0

### Platform

Windos 11

### Printer

Ulti

### Reproduction steps

1. Save 3mf file.
2. Open the 3mf file.
3. Not openned

### Actual results

Not open 3mf file

### Expected results

I want open an 3mf files into Cu

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7488128/cura.log)
The 3mf file:
https://dropmefiles.com/b2msv

sundansx, 2021-11-07:
I am having the same problem after upgrading from 4.80.  It also refuses to open the 3mf 3d model/mesh that represents my bed.  It is one that is included with cura (voron 350mm).  I converted it to a stl and that is opening

fvrmr, 2021-11-08:
Hi @userosos thank you for your report. 
I could import your model with no problems. Does it happen with every 3mf file?
Do you maybe have Chitubox installed? See this issue: https://github.com/Ultimaker/Cura/issues/9668

sundansx, 2021-11-08:
Mine happens with every 3mf file.  Nothing crashes, it just throws up an error dialog as shown.  
I do not have ?Chitubox installed.
I have the same results as the first post.  
I am using Windows 10 21H1, with Cura 4.11.0
I also have Cura 3.5 installed.  I upgraded Cura install from 4.8 to 4.11
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/7498687/cura.log)

nallath, 2021-11-09:
I think you might have the startup optimizer enabled and let it be a bit too aggressive with what should not be loaded. Could you see if disabling it (or enabling a few more files to be loaded) solves it?

fieldOfView, 2021-11-09:
Loading the StartOptimiser plugin would show up in the log, and it doesn't. Also the Start Optimiser plugin does not prohibit the loading of other plugins.

Two plugins that are also notably missing in the log is the `3MFReader` and `3MFWriter` plugins loading. Those are the plugins that ship with Cura that do the loading (and writing) of 3mf files.

It looks like @sundansx has either modified their Cura installation folder, or something very strange happened while installing Cura.

How about @userosos, can you also post a cura.log please, so we can see what is going on on your system?

fieldOfView, 2021-11-09:
Sorry @userosos, you posted a log file already. Both the `3MFReader` and `3MFWriter` plugins are missing in your install too.

nallath, 2021-11-09:
> Sorry @userosos, you posted a log file already. Both the `3MFReader` and `3MFWriter` plugins are missing in your install too.

Yeah, i noticed that too. No idea how that happened. It's as if the entire folder was removed... The logs simply dont mention the 3mf plugins at all.

fieldOfView, 2021-11-09:
Either the entire folder was removed, or there is no `__init__.py` file in the folder.

@userosos and @sundansx, can you check if there is a folder named `3MFReader` in `C:\Program Files\Ultimaker Cura 4.11\plugins` (or whereever you installed the application), and if it contains a file named `__init__.py` ?

sundansx, 2021-11-09:
[image]

They are present...One thing atypical is that I installed cura on another drive (as I did with previous versions) than my boot drive.  This install was also an upgrade from 4.8.0.  I think 3mf worked in that install, but I am not positive.

sundansx, 2021-11-09:
ok, enabling that plugin got my 3mf read/write working.  I was able to load the v350_demon.3mf file from this issue.  I am not sure how that plugin got disabled..and I was not aware that 3mf support was encapsulated in a plugin.  
A suggestion is that there be text included in the error message above to 'check that the plugin is enabled' or 'may be a missing or disabled plugin'.  That would have likely lead me to look there.  The plugin list is quite long, so I did not examine all of them.

Ghostkeeper, 2021-11-12:
Glad to know that worked for you, Sundansx. Was that also the cause of the issue for @userosos?

closed 2021-11-22 by no-response[bot]